polarofficial/polar-ble-sdk #631: Issue after charging Polar 360

mikeread-dev, 2025-06-09:
**Platform on which you observed the bug:** 
- [ X] Android 
- [ ] iOS
- [ ] Other
- [ ] Platform is not relevant for this bug

**SDK version:**
6.3.0
1.2.11 (FW)

**Customer app url**
Link to app store / play store / your website

**Device on which you observed the bug:**
- [ ] Grit X2 Pro
- [ ] Vantage V3
- [ ] Vantage M3
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ X] Polar 360
- [ ] Other
- [ ] Device is not relevant for this bug

**Describe the bug**
On Sunday I put the 360 on the charger for the day, at the end of the day I took it off and tried to sync as normal with my app.  However, part of my sync checks `isFtuDone`.    However, this timed out continuously showing the timeout exception:

```
[info] | 21:56:33 925ms | Checking FTU status (attempt 1)...
[error] | 21:57:56 870ms | Error during FTU status check (attempt 1): Failed to check FTU status: com.polar.androidcommunications.api.ble.model.gatt.client.psftp.BlePsFtpUtils$PftpOperationTimeout: Air packet was not received in required timeline: PlatformException(java.lang.Exception: com.polar.androidcommunications.api.ble.model.gatt.client.psftp.BlePsFtpUtils$PftpOperationTimeout: Air packet was not received in required timeline, com.polar.androidcommunications.api.ble.model.gatt.client.psftp.BlePsFtpUtils$PftpOperationTimeout: Air packet was not received in required timeline, null, null)
```

I tried a number of things to try to fix the issue, but the only thing that worked was a factory reset of the band using the hardware button, when on the charger.  Once I did that the band worked as normal, so it seemed the FW may have got itself in some weird state?

I've charged the band many times without issue, so it's not a consistent bug, the only thing I may have done differently are:
- Left the band on the charger for a long time (maybe 12 hours)
- I did try a sync (connect > stop recording > get offline recordings > delete recordings > restart recording) whilst the band was on the charger, which gave me the expected error of Band Charging.

I can't really offer any more details as I wasn't working, just using the band to test.

rkangast, 2025-06-10:
Hi @mikeread-dev 

The error "Air packet was not received in required timeline" suggests that there are problems with Bluetooth connection. So it is plausible that Polar 360 Bluetooth has entered some problematic state. There are some updates on BT connectivity in firmware version [1.4.38](https://support.polar.com/en/polar-360-firmware-updates). Please, update your device(s).

mikeread-dev, 2025-06-13:
Since I have undated the FW I have not been able to replicate the issue.  Closing the issue.

closed 2025-06-13 by mikeread-dev