polarofficial/polar-ble-sdk #632: Incorrect SleepCycle Data

Sammo98, 2025-06-11:
***Platform your question concerns:*** 
- [ ] Android 
- [x] iOS
- [ ] Other
- [ ] Platform is not relevant for this question

**SDK version:**
6.3

**Customer app url**
N/A

***Device:***
- [ ] Grit X2 Pro
- [ ] Vantage V3
- [ ] Vantage M3
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [x] Polar 360
- [ ] Other
- [ ] Device is not relevant for this question

**Description:**

I'm receiving some strange data for SleepCycle:

```
SleepStartTime: Optional(2025-06-10 23:52:11 +0000)
SleepEndTime: Optional(2025-06-11 07:57:11 +0000)
```

```
Optional([PolarBleSdk.PolarSleepData.SleepCycle(secondsFromSleepStart: Optional(22485), sleepDepthStart: Optional(0.4419035)), PolarBleSdk.PolarSleepData.SleepCycle(secondsFromSleepStart: Optional(24195), sleepDepthStart: Optional(0.8595284)), PolarBleSdk.PolarSleepData.SleepCycle(secondsFromSleepStart: Optional(32445), sleepDepthStart: Optional(0.9919865)), PolarBleSdk.PolarSleepData.SleepCycle(secondsFromSleepStart: Optional(40275), sleepDepthStart: Optional(0.986652)), PolarBleSdk.PolarSleepData.SleepCycle(secondsFromSleepStart: Optional(45525), sleepDepthStart: Optional(0.9691873)), PolarBleSdk.PolarSleepData.SleepCycle(secondsFromSleepStart: Optional(50145), sleepDepthStart: Optional(0.9587177))])
```

I may have messed something up, but this is the raw data from `getSleepData`. Is there a reason why the secondsFromSleepStart does not align with my sleep start and end? The `secondsFromSleepStart` seems correct for the `SleepWakePhase`

```
Optional([PolarBleSdk.PolarSleepData.SleepWakePhase(secondsFromSleepStart: Optional(0), state: Optional(PolarBleSdk.PolarSleepData.SleepWakeState.WAKE)), PolarBleSdk.PolarSleepData.SleepWakePhase(secondsFromSleepStart: Optional(30), state: Optional(PolarBleSdk.PolarSleepData.SleepWakeState.NONREM12)), PolarBleSdk.PolarSleepData.SleepWakePhase(secondsFromSleepStart: Optional(210), state: Optional(PolarBleSdk.PolarSleepData.SleepWakeState.NONREM3)), PolarBleSdk.PolarSleepData.SleepWakePhase(secondsFromSleepStart: Optional(240), state: Optional(PolarBleSdk.PolarSleepData.SleepWakeState.NONREM12)), PolarBleSdk.PolarSleepData.SleepWakePhase(secondsFromSleepStart: Optional(690), state: Optional(PolarBleSdk.PolarSleepData.SleepWakeState.WAKE)), PolarBleSdk.PolarSleepData.SleepWakePhase(secondsFromSleepStart: Optional(870), state: Optional(PolarBleSdk.PolarSleepData.SleepWakeState.NONREM12)), 
```

As a follow up, how does one interpret the `sleepDepthStart`?

Thanks in advance!

pth-pe-gh, 2025-06-24:
@Sammo98 Unfortunately our internal documentation was unclear, so I have escalated this domain expert to clarify the details. Thanks for your patience; we will respond when we have got more information.

chraebslight, 2025-08-28:
Upvote for 

> As a follow up, how does one interpret the sleepDepthStart?

```json
"sleepCycles":[{"secondsFromSleepStart":13515,"sleepDepthStart":0.42340064},{"secondsFromSleepStart":17055,"sleepDepthStart":0.94816226},{"secondsFromSleepStart":25815,"sleepDepthStart":0.9818447},{"secondsFromSleepStart":31815,"sleepDepthStart":0.91918},{"secondsFromSleepStart":35295,"sleepDepthStart":1.0183997}],"sleepEndOffsetSeconds":0,"sleepEndTime":"2025-07-09T06:12:23.000","sleepGoalMinutes":539,"sleepResultDate":"2025-07-09","sleepStartOffsetSeconds":0,"sleepStartTime":"2025-07-09T00:05:53.000",
```

I couldn't find any hints about what this value (type float) indicates.