polarofficial/polar-ble-sdk #639: meanNightlyRecoveryRespirationInterval integer overflow

nbardiuk, 2025-06-27:
**Platform on which you observed the bug:** 
- [ ] Android 
- [x] iOS
- [ ] Other
- [ ] Platform is not relevant for this bug

**SDK version:**
6.3.0

**Customer app url**
Not yet available
endor.global

**Device on which you observed the bug:**
- [ ] Grit X2 Pro
- [ ] Vantage V3
- [ ] Vantage M3
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [x] Polar 360
- [ ] Other
- [ ] Device is not relevant for this bug

**Describe the bug**
`meanNightlyRecoveryRespirationInterval` for nightly recharge looks off with value `2147483647`

Here is a snapshot of nightly recharge detected by the Polar 360 device on firmware version 1.4.38
```json
{
  "ansRate": 4,
  "ansStatus": 3.054100751876831,
  "sdBaselineRRI": 32,
  "meanBaselineRRI": 969,
  "sdBaselineRMSSD": 5,
  "sleepResultDate": "2025-06-20",
  "createdTimestamp": "2025-06-20T03:55:54Z",
  "meanBaselineRMSSD": 43,
  "recoveryIndicator": 5,
  "scoreRateObsolete": 0,
  "meanNightlyRecoveryRRI": 973,
  "meanNightlyRecoveryRMSSD": 43,
  "recoveryIndicatorSubLevel": 41,
  "sdBaselineRespirationInterval": 71,
  "meanBaselineRespirationInterval": 3957,
  "meanNightlyRecoveryRespirationInterval": 2147483647
}
```

**How to Reproduce**
It happened only once so far, no steps to reproduce.

**Expected behavior**
I understand that it's an integer overflow issue.
If respiration detection failed, I would expect to have no value for this metric.

andYgnite, 2025-07-14:
@nbardiuk 

Good to know it is always the same number. 

What is the firmware version on your device?
And which SDK version do you have right now?

nbardiuk, 2025-07-14:
Out of those three cases
* the first was with 6.3.0 SDK on 1.4.38 firmware
* the two newer cases were with 6.4.0 SDK and 2.0.8 firmware

andYgnite, 2025-07-15:
@nbardiuk 

Thanks. Do you have this with iOS only or did you experience with Android SDK also?