polarofficial/polar-ble-sdk #641: Getting steps halts if a day has no step data

viesti, 2025-06-28:
**Platform on which you observed the bug:** 
- [ ] Android 
- [x] iOS
- [ ] Other
- [ ] Platform is not relevant for this bug

**SDK version:**

**Customer app url**
Link to app store / play store / your website

**Device on which you observed the bug:**
- [ ] Grit X2 Pro
- [ ] Vantage V3
- [ ] Vantage M3
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [x] Polar 360
- [ ] Other
- [ ] Device is not relevant for this bug

Firmware version 2.0.8

**Describe the bug**

When calling the PolarActivityApi.getSteps() with a fromDate that doesn't have steps, the whole step retrieval process is aborted. It seems as if the RxSwift Observable returned from PolarActivityUtils.readStepsFromDayDirectory(), when encountering a non-existing folder path, runs the `onError` of the observable from the BlePsFtpClient, but doesn't complete or error-out the observable that is returned to the caller. It seems that no error is returned from the observable either.

I was able to make a query into the past, for days that don't have step data, work with the following patch:

```diff
diff --git a/sources/iOS/ios-communications/Sources/PolarBleSdk/sdk/impl/utils/PolarActivityUtils.swift b/sources/iOS/ios-communications/Sources/PolarBleSdk/sdk/impl/utils/PolarActivityUtils.swift
index 742f660..0c39d1c 100644
--- a/sources/iOS/ios-communications/Sources/PolarBleSdk/sdk/impl/utils/PolarActivityUtils.swift
+++ b/sources/iOS/ios-communications/Sources/PolarBleSdk/sdk/impl/utils/PolarActivityUtils.swift
@@ -34,7 +34,8 @@ internal class PolarActivityUtils {
             .asObservable()
             .subscribe (
                 onError: { error in
-                    BleLogger.error("Failed to list Activity sample files.")
+                    BleLogger.error("Failed to list Activity sample files. \(error)")
+                    emitter(.success(Int(0)))
                 },
                 onCompleted: {
                     var index = 0
```

This way I get a 0 value reported for a day that doesn't have a step directory in the Polar 360.

**How to Reproduce**

Call PolarActivityUtils.getSteps() with a fromDate that is in the past enough to have no step data with a Polar 360 device (using firmware 2.0.8).

**Expected behavior**

Should be able to get step counter value even for day that doesn't have any data.

**Screenshots and logs**

See the description.

viesti, 2025-07-01:
One note from

```diff
-                    BleLogger.error("Failed to list Activity sample files.")
+                    BleLogger.error("Failed to list Activity sample files. \(error)")
+                    emitter(.success(Int(0)))
```

is that this turns a day with no steps into a day with 0 steps, which is different from having no data for that day. I think the idea f skipping such a day and emitting nothing is ok, but somehow I did ran into situation where I had days without data and a day with day, lead to not getting any data at all.

l1va, 2025-07-04:
I think I experienced something similar. My flow was :  getSteps(), deleteData(Activity), getSteps <- stuck here. 
I didn't have time to double check and collect proofs, that is why I skipped opening a ticket, but it sounds important