polarofficial/polar-ble-sdk #643: 24/7 HR samples time shifted

nbardiuk, 2025-07-03:
***Platform your question concerns:*** 
- [ ] Android 
- [x] iOS
- [ ] Other
- [ ] Platform is not relevant for this question

**SDK version:**
6.4.0

**Firmware version:**
2.0.8

**Customer app url**
Endor.global, not yet available

***Device:***
- [ ] Grit X2 Pro
- [ ] Vantage V3
- [ ] Vantage M3
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [x] Polar 360
- [ ] Other
- [ ] Device is not relevant for this question

**Description:**

We record both offline HR data and 24/7 HR samples. And for users, we have noticed a 2-hour shift in 24/7 HR sample compared to offline HR data. Users are in the Europe/Oslo timezone, which is 2 hours ahead of UTC. 
It is worth noting that this behavior occurs in users on the day they update firmware from 1.4.38 to 2.0.8.

[image]
Purple - offline HR data
Grey - 24/7 HR with timed trigger
Orange - 24/7 HR with high activity trigger

What could explain this shift that we see between the two sources of HR data?

nbardiuk, 2025-07-03:
@jimmyzumthurm 

We have several users who updated to firmware 2.0.8, and only two users are experiencing this time shift. The time shift started around the time they updated the firmware, and didn't resolve automatically the next day.

Yes, the difference between local time and UTC can explain the 2-hour time difference between sources.

When we set the device's `setLocalTime`, it's always UTC timezone, and we read data with the same assumption. It might be that the device's local time is not in UTC, but the local time from the iPhone. I've just checked that SDK sets local time using iPhone timezone during first use, and I think those users also had reset their polar devices around that time https://github.com/polarofficial/polar-ble-sdk/blob/3b3f0164da85eebb3b480f2b59957b49323e0c57/sources/iOS/ios-communications/Sources/PolarBleSdk/sdk/impl/PolarBleApiImpl.swift#L2404

nbardiuk, 2025-07-03:
I will ask users to set the time on their devices through the app again. And come back to report if setting the local time to UTC resolves the drift we observe.

viesti, 2025-07-04:
One thing that makes me nervous is the case of a user traveling with a device. For example, if user travels to another country that has a different timezone, and while there, if the application would  call setLocalTime() with timezone in the travel destination. Then when user travels back, and if setLocaltTime() is not called for some time, the device would get left in travel timezone.

For situations like these we've been thinking that it would be safer to have device always follow UTC. But I'm not sure if this assumption causes problems with assumptions made by the SDK.

nbardiuk, 2025-07-08:
We have confirmed that on devices where users experience a time shift between offline HR and 24/7 samples, calling `setLocalTime` through our app again with UTC timezone resolves the issue, and both sources report the same time.

After reading your reply that sleep data also uses local time, I've checked that only 24/7 HR samples were affected. The sleep data was okay - the sleep start and end times aligned with the low HR levels from the offline HR recordings.

I don't know where the root cause of the problem is. I'm not 100% sure what happened when users started experiencing the shift - the first time use is only an assumption, a smoking gun.

viesti, 2025-09-18:
Sorry for asking, but @orestesgaolin did you experience anything similar to this, by any chance?

viesti, 2025-09-19:
Thank you @orestesgaolin ! 

I wish Swift/Objc would have JSR-310 :) I'm not sure if there is LocalDate equivalent in Swift, it's a bit same dilemma as with JavaScript.

And I also agree that my gut says that labeling with UTC, and leaving timezone conversions as "UI problem" (for example when daylight savings change), would be better, or so my gut says, reality might be more problematic :D.

Will have to dig into the Swift SDK a bit more.