polarofficial/polar-ble-sdk #647: Does sleep detection use timezone of setLocalTime

viesti, 2025-07-05:
***Platform your question concerns:*** 
- [ ] Android 
- [x] iOS
- [ ] Other
- [ ] Platform is not relevant for this question

**SDK version:**

SDK: 6.4.0
Firmware: 2.0.8

**Customer app url**
Not yet released
***Device:***
- [ ] Grit X2 Pro
- [ ] Vantage V3
- [ ] Vantage M3
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [x] Polar 360
- [ ] Other
- [ ] Device is not relevant for this question

**Description:**

We are currently using UTC as the timezone with setLocalTime(). Idea was to try avoid timezone related issues, for example if user is travelling (say, user travels, application calls setLocalTime() in destination timezone, user travels back, but setLocalTime() does not get called and device is left in wrong timezone).

We are also using sleep data.

What I'd like to ask is that does the sleep detection procedure use timezone information. FTU configuration has sleepGoalMinutes, but it does not have got to bed and wake up time.

Also, general questions is that should setLocalTime() use phone-local timezone instead of UTC. Is use of UTC problematic, or does it run against assumptions done by the SDK or device firmware?

viesti, 2025-09-04:
Hi! @jimmyzumthurm 

Maybe bad thread to ask, but does the device user's local timezone when creating day directories for step data?

I'm asking since if user travels, if the step day directories are timezone aware, then if user's timezone changes, the SDK reading code uses a timezone-aware formatter in

```
// Timezone aware formatter
private let dateFormat: DateFormatter = {
    let formatter = DateFormatter()
    formatter.dateFormat = "yyyyMMdd"
    formatter.locale = Locale(identifier: "en_US_POSIX")
    return formatter
}()
private let TAG = "PolarActivityUtils"

internal class PolarActivityUtils {
    
    /// Read step count for given date.
    static func readStepsFromDayDirectory(client: BlePsFtpClient, date: Date) -> Single<Int> {
        BleLogger.trace(TAG, "readStepsFromDayDirectory: \(date)")
        return Single<Int>.create { emitter in
            let activityFileDir = "\(ARABICA_USER_ROOT_FOLDER)\(dateFormat.string(from: date))/\(ACTIVITY_DIRECTORY)" // Use of the formatter
```
then it means that if the folder in firmware was created by formatting based on different timezone than at time of query, we might end up reading wrong date folder.

pth-pe-gh, 2025-11-13:
@viesti - I noticed your latest question from Sep 4 at least was not responded to yet. Looking into it now, I notice also that I don't fully understand your question or rather, what is the potential problem. Could you elaborate how the SDK based app could end up reading wrong date folder?