polarofficial/polar-ble-sdk #649: Can't connect to my 360 after a factory reset.

ymatt345, 2025-07-07:
**Platform on which you observed the bug:** 
- [ ] Android 
- [ ] iOS
- [ ] Other
- [ ] Platform is not relevant for this bug

**SDK version:**

**Customer app url**
Link to app store / play store / your website

**Device on which you observed the bug:**
- [ ] Grit X2 Pro
- [ ] Vantage V3
- [ ] Vantage M3
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [✓] Polar 360
- [ ] Other
- [ ] Device is not relevant for this bug

I've been using my app to test with the 360 for about 4 months now.

Recently, the 360's BLE connection has been dropping out again and again. I was able to use the 360 through the app's recovery procedure for a while, but it eventually dropped out again.

So I did a factory reset and tried to reconnect the 360, and although it showed "DEVICE CONNECTING: (Device ID: "E462542B", Address: 99E6F0CA-D36E-5674-73BE-BDBFFF29E09C, RSSI: -42, Name: "Polar 360 E462542B", Connectable: true)", it didn't show "DEVICE CONNECTED:" and the connection failed.

This symptom seems to be because it remembers the problematic connected device even after the factory reset and refuses to connect.

So I did a similar test on this 360 with a different iPhone app, the connection after the above process was successful, but after the factory reset again, and tried to reconnect the 360, it stopped at "Device connecting:" as before, and could not proceed any further.

Given this situation, it appears that any connected mobile devices that are already stored in the 360 ​​will be remembered and will not be able to be reconnected if you perform a factory reset.

It is easy to duplicate this symptom, so please test it to confirm.
Performing a factory reset will prevent the device from being used on your previous mobile device, so the impact of this issue is significant and the action will be necessary.

ymatt345, 2025-07-11:
@jimmyzumthurm 
I have sent back to the 360 to customer support in Japan, so I am not able to verify your suggestion.
Will simply remove "Forget thsi device" really fix this issue?

jimmyzumthurm, 2025-07-11:
@ymatt345 It should fix the issue. When the device is in factory defaults state there is no reason why it would refuse connection and pairing from mobile, unless the mobile tries to re-establish the security with invalid keys.

ymatt345, 2025-07-12:
@jimmyzumthurm 
Can you duplicate the probllem by doing Factory reset tha has BLE connection with iPhone and just try device connection again.  This should result in 360 to"Device connecting" status.
And try connection again after  manually setting "Forget this device" for BLE setting. 
Please let me know if you can dupplicate the issue and solve the issue by manuallt setteing "Forget this device", and inform the solution for this issue to your ww support center .
Best regards,