polarofficial/polar-ble-sdk #650: Accelerometer values clipped at +-2000mG for higher sampling frequencies

sunukpfr, 2025-07-10:
***Platform your question concerns:*** 
- [ ] Android 
- [ ] iOS
- [ ] Other
- [x ] Platform is not relevant for this question

**SDK version:**

**Customer app url**
Link to app store / play store / your website

***Device:***
- [ ] Grit X2 Pro
- [ ] Vantage V3
- [ ] Vantage M3
- [ ] Polar OH1
- [x ] Polar Verity Sense
- [x ] Polar H10
- [ ] Polar H9
- [ ] Polar 360
- [ ] Other
- [ ] Device is not relevant for this question

**Description:**
It seems as if at fs=200hz the accelerometer values are clipped at +-2000mG. Is there any way around this?

jimmyzumthurm, 2025-07-11:
Hi @sunukpfr , 
Are you configuring the range that you pass as settings when starting the measurement ?

sunukpfr, 2025-07-11:
Hi @jimmyzumthurm 
Thank you for your comment. Thought I did but turns out I didn't. Can't fathom how I made the same mistake 2 times in a row, but looks like I have. Pilot error.