polarofficial/polar-ble-sdk #656: Polar 360 temperature streaming does not work

tsukinami7, 2025-07-16:
**Platform on which you observed the bug:** 
- [x ] Android 
- [ ] iOS
- [ ] Other
- [ ] Platform is not relevant for this bug

**SDK version:**
6.4.0
firmware 1.2.11 > 2.0.8

**Device on which you observed the bug:**
- [ ] Grit X2 Pro
- [ ] Vantage V3
- [ ] Vantage M3
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ x] Polar 360
- [ ] Other
- [ ] Device is not relevant for this bug

**Describe the bug**
When I call the following code, it returns an error 'com.polar.androidcommunications.api.ble.exceptions.BleControlPointCommandError: pmd cp command GET_MEASUREMENT_SETTINGS error: ERROR_NOT_SUPPORTED'.

```
private lateinit var polarAPI: PolarBleApi
---

polarAPI.requestStreamSettings(identifier, PolarBleApi.PolarDeviceDataType.TEMPERATURE)
                            .toFlowable()
                            .flatMap { sensorSetting: PolarSensorSetting -> polarAPI.startTemperatureStreaming(identifier, sensorSetting.maxSettings()) }
                            .observeOn(AndroidSchedulers.mainThread())
                            .subscribe(
                                { data: PolarTemperatureData ->
                                    val temp: Double? = data.samples.firstOrNull()?.temperature?.toDouble()
                                    if (temp != null) {
                                        Log.d("PolarDebug", "Tempdata: $temp")
                                    }
                                }, { error ->
                                    Log.e("PolarDebug", "Temperature streaming error: $error")
                                }
```

Both the 'deviceConnecting' and 'deviceConnected' callbacks were properly called. 

I successfully used the same code on 360 firmware version 1.2.11 last week.
I therefore believe that the temperature streaming feature is affected by firmware version changes.

tsukinami7, 2025-07-16:
Wow! Thank you so much!!
After switching the data type from `temp` to `skinTemp`, it started working properly!
Just a TIPS for everyone: when working with the temperature stream, you need to use `startSkinTemperatureStreaming` , but note that the returned data type is not PolarSkinTemperatureData — it's PolarTemperatureData.
I'll share the working kotlin code below for reference.

```
polarAPI.requestStreamSettings(identifier, PolarBleApi.PolarDeviceDataType.SKIN_TEMPERATURE)
                            .toFlowable()
                            .flatMap { sensorSetting: PolarSensorSetting -> polarAPI.startSkinTemperatureStreaming(identifier, sensorSetting.maxSettings()) }
                            .observeOn(AndroidSchedulers.mainThread())
                            .subscribe(
                                { data: PolarTemperatureData ->
                                    val temp: Double? = data.samples.firstOrNull()?.temperature?.toDouble()
                                    if (temp != null) {
                                        Log.d("PolarDebug", "Tempdata: $temp")
                                    }
                                }, { error ->
                                    Log.e("PolarDebug", "Temperature streaming error: $error identifier $identifier")
                                }
                            )
```

jimmyzumthurm, 2025-07-16:
Hi, @l1va yes that is correct. Device used to use "temp" and it was changed to "skinTemp". The reason is that the "temp" type was originally thought to communicate the "device" temperature (even if I'm not sure any of our products are using this), so a new skinTemp type was created so that a device could expose both types at the same time.