polarofficial/polar-ble-sdk #669: Question: What is usb connection mode?

janbarari, 2025-07-29:
Platform: Android
Device: Polar 360
SDK Version: 6.4.0
Firmware: 2.0.8
Android Version: 15

Dear Polar Team,

Can you please explain what USB connection mode does and whether it's related to the Polar360 device?

Thank you for your time and consideration.

Best regards,
Mehdi from Virgobit

pth-pe-gh, 2025-08-01:
USB connection mode refers to connecting the sensor to another device using USB cable for communications. SDK allows setting this feature on and off. Polar 360 supports this feature in version 2.0.8, and the connection mode is off by default.

janbarari, 2025-08-01:
So instead of using Bluetooth, we can establish a USB connection between the Android device and the sensor?

pth-pe-gh, 2025-08-04:
USB is not supported by BLE SDK. USB is currently utilised by other tools. For consumers, Polar FlowSync is a computer program that connects to Polar devices using USB and allows syncing one's personal training data from Polar devices to Polar Flow account. USB connections may be also used by Polar service technicians in customer care cases.

janbarari, 2025-08-04:
Thanks @pth-pe-gh
I close the ticket.