polarofficial/polar-ble-sdk #670: The bleSdkFeatureReady function from the SDK is not triggered

amart384, 2025-07-30:
**Platform on which you observed the bug:** 
- [X ] Android 
- [ ] iOS
- [ ] Other
- [ ] Platform is not relevant for this bug

**SDK version:**
5.5.0
6.4.0

**Customer app url**
https://github.com/polarofficial/polar-ble-sdk/tree/master/examples/example-android/androidBleSdkTestApp

**Device on which you observed the bug:**
- [ ] Grit X2 Pro
- [ ] Vantage V3
- [ ] Vantage M3
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ X] Polar H10
- [ ] Polar H9
- [ ] Polar 360
- [ ] Other
- [ ] Device is not relevant for this bug

**Describe the bug**
The bleSdkFeatureReady function is not triggered when multiple applications are running. Therefore, it is not possible to start the ECG stream.
Tested applications are AndroidBleSdkTestApp (with SDK) and the official Polar Beat application.

**How to Reproduce**
Open the Polar Beat application
Start recording/capturing
Open the AndroidBleSdkTestApp (with the SDK)
Press the AUTO CONNECT button (it connects successfully)
Start recording/capturing using the START ECG STREAM button

The following exception is thrown:
com.polar.androidblesdk              E  ECG stream failed. Reason com.polar.sdk.api.errors.PolarDeviceNotFound

**Expected behavior**
The function bleSdkFeatureReady is not triguered, therefore, it is not possible to start the ECG stream.
bleSdkFeatureReady(@NonNull final String identifier,
                               @NonNull final PolarBleApi.PolarBleSdkFeature feature)

**Screenshots and logs**
The following is triggered:
public void deviceConnected(@NonNull PolarDeviceInfo polarDeviceInfo)

amart384, 2025-08-25:
# Bug Report: PSDC App does not trigger `getAvailableOnlineStreamDataTypes`

## Description

The PSDC App connects, but the function `getAvailableOnlineStreamDataTypes` is not triggered.

PolarBleApi.PolarBleSdkFeature.FEATURE_POLAR_ONLINE_STREAMING -> {
    api.getAvailableOnlineStreamDataTypes(identifier)
        .subscribe(
            { types ->

## Steps to Reproduce

1. Close the PSDC App  
2. Open the PolarBeat App  
3. Start a recording with PolarBeat  
4. Start the PSDC App  
   - PSDC connects but does not allow starting the recording  
   - The UI does not show the button to start recording (screenshot attached)  
5. Exit the PSDC App  
6. The recording in PolarBeat stops  
7. Start the PSDC App again  
   - Now it can connect and start the recording  
   - The function `getAvailableOnlineStreamDataTypes` is called  

## Expected Behavior

- PSDC should connect and allow starting the recording regardless of whether PolarBeat was previously active.

## Actual Behavior

- PSDC connects but does not trigger `getAvailableOnlineStreamDataTypes`.  
- The recording button does not appear in the UI.

## Attachments

- PSDC App logcat when it connects but **cannot** start recording:  
  `connected_cannot_stream.log`  

- PSDC App logcat when it connects and **can** start recording:  
  `connected_can_stream.log`  

- Screenshot: 

[connected_can_stream.log](https://github.com/user-attachments/files/21976908/connected_can_stream.log)
[connected_cannot_stream.log](https://github.com/user-attachments/files/21976909/connected_cannot_stream.log)
[image]

Thank you !

amart384, 2025-08-27:
Hi, Thank you for your feedback!

The issue has been tested on Android 12 and 15.
Yes, the option for 2 Bluetooth devices was enabled.
By the way, your sensor’s firmware is a very old version. I tested it with the latest one.

I didn’t mention earlier that both applications are running on the same phone.

I don’t think the problem is when two apps (on the same phone) access the HR.

The problem happens when an app using the SDK tries to **stream the ECG** while Polar Beat has already started a recording. It also only occurs in this sequence: first start a recording with Polar Beat, then with PSDC.

The same problem also appears if PolarFlow is used. These combinations have been tested:

Polar Beat & AndroidBleSdkTestApp

Polar Beat & PSDC

Polar Flow & PSDC

PSDC & AndroidBleSdkTestApp

The reason why solving this issue is important: imagine a user starts Polar Beat, forgets to stop the recording, and removes the sensor. The Polar Beat notification disappears. After some time, the user puts the sensor back on to start a recording with an SDK-based app — but Polar Beat automatically starts first. Then the user tries to start the SDK app **(for ECG streaming)**, but it doesn’t work. The SDK app connects but never receives _bleSdkFeatureReady_, **so the SDK app cannot work**.

Sometimes, users also want to keep both apps active (Polar Beat and an SDK-based app) in order to compare performance, I assume.

If the problem cannot be solved, it would still be very helpful to at least capture this situation with a _bleSdk**Not**FeatureReady_ event, so the user can be informed to close one of the apps.

amart384, 2025-09-06:
Hi again,

The app in internal test on (google play store) generates these exceptions on diferents devices. No further information is available. Perhaps some of these exception traces may provide some clues about the previously described issue.

Perhaps NullPointerException exceptions are easy to check/handle to avoid API crash.

- com.polar.androidcommunications.enpoints.ble.bluedroid.host.BDDeviceListenerImpl$2.scanStartError
com.polar.androidcommunications.api.ble.exceptions.BleStartScanError

- com.polar.androidcommunications.api.ble.model.gatt.BleGattBase.lambda$waitNotificationEnabled$1
com.polar.androidcommunications.api.ble.exceptions.BleAttributeError

- com.polar.androidcommunications.enpoints.ble.bluedroid.host.BDDeviceListenerImpl$1.setPhy
java.lang.NullPointerException

- com.polar.androidcommunications.api.ble.model.gatt.client.pmd.BlePMDClient.readFeature$lambda$18
java.lang.Exception

- com.polar.androidcommunications.enpoints.ble.bluedroid.host.GattCallback.onServicesDiscovered
java.lang.NullPointerException

Thank you.

palmqvisti, 2025-11-24:
Hi @amart384, could you try the Polar Sensor Data Collector app to test streaming both HR and ECG?
https://github.com/polarofficial/polar-ble-sdk/tree/master/examples

rkangast, 2026-01-30:
Hi @amart384 

Sorry for this late response. The issue has been attended to. Please update to newest SDK version and try out. I will now close this ticket, but if the problem persist please feel free to re-open.

amart384, 2026-02-02:
Hello,
The problem persist. It has been tested with version 6.14.0

It works if PSDC is started first and then Polar Beat.

It does NOT work if Polar Beat is started first and then PSDC.

Thank you!

amart384, 2026-02-13:
Hi @rkangast 
Thank you very much for your time and effort investigating this issue.

I personally can work around the issue without major problems.

However, the challenge is that other users of the application are not aware that they need to start our app first before using Polar Beat. This may lead to confusion.

Thank you again.