polarofficial/polar-ble-sdk #676: Get timestamp from PolarHrData?

abbjetmus, 2025-08-08:
***Platform your question concerns:*** 
- [ ] Android 
- [ ] iOS
- [ ] Other
- [ X] Platform is not relevant for this question

**SDK version:**

**Customer app url**
Link to app store / play store / your website

***Device:***
- [ ] Grit X2 Pro
- [ ] Vantage V3
- [ ] Vantage M3
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ X] Polar 360
- [ ] Other
- [ X] Device is not relevant for this question

**Description:**
Hi! I would like to request timestamp on PolarHrData. Before we got an update with timestamp on PolarPpiData, but it would be nice to have it on all data objects. Although our use case requires at this time on on PolarHrData. Is it possible to get this in next release?

Offline data of PolarHrData does not contain timestamp.

https://github.com/polarofficial/polar-ble-sdk/issues/558

samulimaa, 2025-08-19:
Hello, good question.

Reason why PolarHrData does not contain timestamp is due to fact that timestamps are not part of the [BLE Heart Rate Service GATT specification](https://www.bluetooth.org/docman/handlers/downloaddoc.ashx?doc_id=239866):

> 
>     3.4 Requirements for Time-Sensitive Data
> 
>     The Heart Rate Measurement characteristic contains time sensitive data and is considered a time-sensitive characteristic, thus the following requirements apply:
> 
>     Since this service does not provide for a time stamp to identify the measurement time (and age) of the data, the value of the Heart Rate Measurement characteristic shall be discarded if either the connection does not get established or if the notification is not successfully sent to the Client (e.g., link loss).

You could try to implement timestamps on your app using current time + 1s for every sample

shahshahani-ee, 2025-08-26:
@samulimaa Any further explanations?

abbjetmus, 2025-08-26:
@shahshahani-ee When you store PPI offline, the HR calculations differ from the HR values contained in the PPI data. During motion, those values are always 0 and therefore not useful. However, the HR values stored separately are much more reliable.

https://github.com/polarofficial/polar-ble-sdk/issues/411

palmqvisti, 2025-11-25:
Hi @abbjetmus, @shahshahani-ee, simply removing the device does not mess the sample time calculation, because samples with HR = 0 are added to data when device is not worn and HR is not detected. Here is an example file attached to demonstrate this. The recording is 178 seconds long in total and having a non-wear period in the middle. The export contains 178 samples and non-wear time is indicated as HR=0 samples.

[Offline_recording_HR_2025-11-25_10-38-19.txt](https://github.com/user-attachments/files/23742375/Offline_recording_HR_2025-11-25_10-38-19.txt)