polarofficial/polar-ble-sdk #678: getSteps error when no data

Marcus122, 2025-08-12:
**Platform on which you observed the bug:** 
- [X] Android 
- [ ] iOS
- [ ] Other
- [ ] Platform is not relevant for this bug

**SDK version:**
6.4.0

**Customer app url**
Link to app store / play store / your website

**Device on which you observed the bug:**
- [ ] Grit X2 Pro
- [ ] Vantage V3
- [ ] Vantage M3
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [X] Polar 360
- [ ] Other
- [ ] Device is not relevant for this bug

**Describe the bug**
Description of your observation

We are seeing an issue when calling the `getSteps` API method. There is an error being thrown when there are no steps for that day eg band hasn't been worn. 

Looking at PolarActivityUtils.kt -> listFiles I would guess it's trying to look for files that do not exist?

It's causing a bit of problem for us as I don't think we can catch the error and so our app just crashes.

```
io.reactivex.rxjava3.exceptions.OnErrorNotImplementedException: The exception was not handled due to missing onError handler in the subscribe() method call. Further reading: https://github.com/ReactiveX/RxJava/wiki/Error-Handling | java.lang.Exception: com.polar.androidcommunications.api.ble.model.gatt.client.psftp.BlePsFtpUtils$PftpResponseError: Request failed: Error: 103
```

```
Caused by: java.lang.Exception: com.polar.androidcommunications.api.ble.model.gatt.client.psftp.BlePsFtpUtils$PftpResponseError: Request failed:  Error: 103
	at com.polar.sdk.impl.utils.PolarActivityUtils$listFiles$2.apply(PolarActivityUtils.kt:210)
	at com.polar.sdk.impl.utils.PolarActivityUtils$listFiles$2.apply(PolarActivityUtils.kt:209)
```

**How to Reproduce**
If you know how the bug may be reproduced please add the steps here

Do not wear the polar device for a day then the next day call `getSteps` passing in yesterdays date.

**Expected behavior**
Description of what you expected to happen. 

**Screenshots and logs**
If applicable, add screenshots or logs to help explain your problem.

gmbeard, 2025-08-19:
@samulimaa FWIW, I'm able to prevent this exception using the following patch...

```diff
diff --git a/sources/Android/android-communications/library/src/sdk/java/com/polar/sdk/impl/utils/PolarActivityUtils.kt b/sources/Android/android-communications/library/src/sdk/java/com/polar/sdk/impl/utils/PolarActivityUtils.kt
index 7a3230e..087ce03 100644
--- a/sources/Android/android-communications/library/src/sdk/java/com/polar/sdk/impl/utils/PolarActivityUtils.kt
+++ b/sources/Android/android-communications/library/src/sdk/java/com/polar/sdk/impl/utils/PolarActivityUtils.kt
@@ -206,8 +206,8 @@ internal object PolarActivityUtils {
             condition = condition)
             .map {
                 it.first
-            }.onErrorResumeNext { throwable: Throwable ->
-                Flowable.error(Exception(throwable))
+            }.onErrorResumeNext { _: Throwable ->
+                Flowable.empty()
             }
     }
```

rkangast, 2025-09-09:
This has been fixed in release [6.6.0](https://github.com/polarofficial/polar-ble-sdk/tree/6.6.0)