polarofficial/polar-ble-sdk #685: Offline recording skin temperature parse failure with 3.0.16

viesti, 2025-08-30:
**Platform on which you observed the bug:** 
- [ ] Android 
- [x] iOS
- [ ] Other
- [ ] Platform is not relevant for this bug

**Platform version:**
iOS 18.6.2

**SDK version:**
6.5.0

**Customer app url**
https://apps.apple.com/app/id123456789

**Device on which you observed the bug:**
- [ ] Grit X2 Pro
- [ ] Vantage V3
- [ ] Vantage M3
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [x] Polar 360
- [ ] Other
- [ ] Device is not relevant for this bug

**Describe the bug**

I saw a single failing offline skin temperature file when upgrading from firmware 2.0.8 to 3.0.16 with error like this:

```
gattDataError(description: "Timestamp delta cannot be calculated for the frame, because previousTimeStamp 809814611247243652 is bigger than timestamp 809814526368930664")
```

Device seems to record offline skin temperature still, but I'm not sure how such bad data got created. 

File has metadata

```
{:size 2330,
  :date #inst "2025-08-29T17:26:12.000-00:00",
  :type "skinTemperature",
  :path "/U/0/20250829/R/202612/SKINTEMP0.REC"}
```

I did the firmware update earlier in the same day (IIRC).

It would also be nice to log the actual parsing error, now I had too debug to catch the underlying error at https://github.com/polarofficial/polar-ble-sdk/blob/master/sources/iOS/ios-communications/Sources/PolarBleSdk/sdk/impl/PolarBleApiImpl.swift#L1548:

```
                                          let requestResult =
                                               Single.deferred { client.request(request) }
                                              .map { dataResult -> OfflineRecordingData<Any> in
                                                  do {
                                                      let pmdSecret = try secret.map { try PolarDataUtils.mapToPmdSecret(from: $0) }
                                                      return try OfflineRecordingData<Any>.parseDataFromOfflineFile(
                                                          fileData: dataResult as Data,
                                                          type: PolarDataUtils.mapToPmdClientMeasurementType(from: entry.type),
                                                          secret: pmdSecret,
                                                          lastTimestamp: lastTimestamp
                                                      )
                                                  } catch {
                                                      NSLog("OFFLINE RECORD PARSE FAILURE: \(error)") // <<<---- DEBUG
                                                      throw PolarErrors.polarOfflineRecordingError(description: "Failed to parse data")
                                                  }
                                              }
```

viesti, 2025-08-30:
Hmm, timestamps seem from 1970's, maybe device clock got reseted while recording was running.

rkangast, 2025-09-02:
Did you have skin temperature measurement going on when starting the upgrade. That might lead to some unexpected behaviour. So your assumption is probably correct as the clock gets set up during FWU after the device has been first factory reseted (resets clock too, which explains the 1970 date).