polarofficial/polar-ble-sdk #689: getSteps time handling issue

viesti, 2025-09-04:
**Platform on which you observed the bug:** 
- [ ] Android 
- [x] iOS
- [ ] Other
- [ ] Platform is not relevant for this bug

**Platform version:**
iOS

**SDK version:**
6.5.0

**Device on which you observed the bug:**
- [ ] Grit X2 Pro
- [ ] Vantage V3
- [ ] Vantage M3
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [x] Polar 360
- [ ] Other
- [ ] Device is not relevant for this bug

**Describe the bug**

The `getSteps()` API is implemented by making a list of dates for reading the data from the firmware.

The implementation takes the `fromDate` parameter, and increments if by 1 day, while comparing it to <= `toDate` parameter.

If the `fromDate` has a time part that is in the evening, and the `toDate` has time part in the morning, since `fromDate` is incremented by one full day, this leads to skipping the `toDate` in such a case, since on the upper edge, the `fromDate` time part will be in the evening, which will be larger than the `toDate` part.

Also, the return values all use the time part of that start date, so for user of the API, it is a bit confusing to get values per each day, having time part of the `fromDate`.

I think there is a bit confusion, since the implementation reads day directories and sums up array of values (binned by minute, maybe?), so actually, the current API is day-based only, so it would make sense reflect that in the API too.

I think the [API proposal in issue 627](https://github.com/polarofficial/polar-ble-sdk/issues/627#issuecomment-3205488074) would help this situation. I think having time in that start/end parameters is a bit problematic, if data is binned counter values.

rkangast, 2025-09-05:
Hi @viesti 

The new API described in issue #627 will is aimed to be released in the next SDK release (6.6.0) on Wednesday 10th September.

viesti, 2025-09-05:
Nice! Thank you :)

closed 2025-10-07 by palmqvisti