polarofficial/polar-ble-sdk #692: 360 disconnects after reconnection following doRestart()

viesti, 2025-09-09:
**Platform on which you observed the bug:** 
- [ ] Android 
- [x] iOS
- [ ] Other
- [ ] Platform is not relevant for this bug

**Platform version:**
iOS 18.6.2

**SDK version:**
SDK: 6.5.0
Firmware: 2.0.8

**Customer app url**
https://apps.apple.com/fi/app/endor-global/id6744930227

**Device on which you observed the bug:**
- [ ] Grit X2 Pro
- [ ] Vantage V3
- [ ] Vantage M3
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [x] Polar 360
- [ ] Other
- [ ] Device is not relevant for this bug

**Describe the bug**

When issuing `doRestart()` the device does connect back to the phone, but disconnect after few seconds.

In our connection procedure, we check if FTU is done, and the device disconnection seems to happen right after the FTU check.

We now have a fallback where after calling doRestart(), we set a timer in the app to connect back to the device.

The initial reason to call doRestart() was that we check the clock of the device and also have a patch for the SDK to read the timezone configured to the device. If the phone timezone differs from device timezone, then we call setLocalTime() with the current phone timezone, and as a precaution, call doRestart(), to make sure that ongoing recording would use the new timezone, so device is not left on wrong timezone.

But this seems to happen also after plain doRestart() call.

It might be that we have something specific in the connection handling, but wanted to report this still, if there are known issues around doRestart().

**How to Reproduce**
Call doRestart() and wait for device to connect back.

**Expected behavior**
Expecting device to maintain connection that it does after connecting back after restart is done.

pth-pe-gh, 2025-09-09:
@viesti thanks for the report. Since we have new SDK release in release pipeline at the moment, could you kindly verify if this issue persists after upgrading to new SDK version 6.6.0, which should become available later this week..

viesti, 2025-09-09:
Nice!

I'm a bit busy now, but will ping back when I get to test drive 6.6.0 :)

palmqvisti, 2025-10-07:
Hi @viesti, did you have time test this with the newer SDK versions 6.6.0 or 6.7.0?