polarofficial/polar-ble-sdk #698: Polar360 getActiveTime not updating after SDK update

Marcus122, 2025-09-15:
**Platform on which you observed the bug:** 
- [X] Android 
- [ ] iOS
- [ ] Other
- [ ] Platform is not relevant for this bug

**Platform version:**

**SDK version:**
6.6.0

**Customer app url**
Link to app store / play store / your website

**Device on which you observed the bug:**
- [ ] Grit X2 Pro
- [ ] Vantage V3
- [ ] Vantage M3
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [X] Polar 360
- [ ] Other
- [ ] Device is not relevant for this bug

**Describe the bug**
I've just updated to the latest SDK version 6.6.0 and there seems to be an issue with getActiveTime not updating.

If I run getActiveTime on SDK version 6.4.0 I get 
```
timeSedentary: {
    seconds: 0,
    millis: 0,
    minutes: 2,
    hours: 2
},
timeLightActivity: {
    seconds: 30,
    millis: 0,
    minutes: 45,
    hours: 0
}
```
After updating to 6.6.0 these values never change throughout the day.

**How to Reproduce**
Make sure there is activity data when using SDK 6.4.0. Then upgrade to SDK 6.6.0 and call getActiveTime after a period of time - the values don't seem to have changed.

**Expected behavior**
For getActiveTime to keep updating once the SDK has been updated.

**Screenshots and logs**
If applicable, add screenshots or logs to help explain your problem.

rkangast, 2025-09-17:
Yes the device collects data in non-stop manner but does not flush the data immediately to the DSUM file.

The DSUM file is created (if missing) and data flushed to the file upon:
- [sendInitializationAndStartSyncNotifications](https://github.com/makevisible/polar-ble-sdk/blob/master/sources/Android/android-communications/library/src/sdk/java/com/polar/sdk/api/PolarBleApi.kt#L528) method call to the device
-  At midnight
-  When date changes on device. 

So please try call sendInitializationAndStartSyncNotifications method. (And remember to call sendTerminateAndStopSyncNotifications method afterwards.)

Marcus122, 2025-09-17:
@rkangast I've tried calling those methods and still get the same error in the logs.

We've downloaded and ran the sample ios app in this repo on a factory reset polar360 and are also getting the same result.

Is there anything else we can try?

Thanks

Marcus

rkangast, 2025-09-18:
Hello,

It seems that there is a bug in "Get Active Time" feature in iOS example app and it does not show the active time values as it should. This will be fixed upon next SDK release.

Please, do try with [Android example app](https://github.com/polarofficial/polar-ble-sdk/tree/master/examples/example-android/polar-sensor-data-collector). I apologise for this inconvenience.