polarofficial/polar-ble-sdk #699: Allow to read timezone written via setLocalTime()

viesti, 2025-09-16:
***Platform your question concerns:*** 
- [ ] Android 
- [ ] iOS
- [ ] Other
- [x] Platform is not relevant for this question

**SDK version:**
6.5.0
firmware 3.0.31

**Customer app url**
https://apps.apple.com/fi/app/endor-global/id6744930227

***Device:***
- [ ] Grit X2 Pro
- [ ] Vantage V3
- [ ] Vantage M3
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [x] Polar 360
- [ ] Other
- [x] Device is not relevant for this question

**Description:**

The `[setLocalTime](https://polarofficial.github.io/polar-ble-sdk/polar-sdk-ios/Protocols/PolarBleApi.html#/s:11PolarBleSdk0aB3ApiP12setLocalTime_4time4zone7RxSwift17PrimitiveSequenceVyAG16CompletableTraitOs5NeverOGSS_10Foundation4DateVAO0G4ZoneVtF)` API takes a Date and a Timezone.

The current Date can be read with `[getLocalTime](https://polarofficial.github.io/polar-ble-sdk/polar-sdk-ios/Protocols/PolarBleApi.html#/s:11PolarBleSdk0aB3ApiP12getLocalTimey7RxSwift17PrimitiveSequenceVyAE11SingleTraitO10Foundation4DateVGSSF)`, but there is no way to read the TimeZone written by setLocalTime.

Currently, we have a patch of our own, which is basically copy of getLocalTime, but which returns the TimeZone.

I think it is necessary to read device timezone, for example in cases where user travels to difference timezone (or timezone gets changed for other reasons).

samulimaa, 2025-09-16:
Hi, thanks for reporting this, I have created an internal ticket for implementation. TC-65647

Just FYI that in Android side reading timezone is possible as returned Calendar object contains it.

viesti, 2025-09-16:
Aaa, bummer that my project isnon iOS :D