polarofficial/polar-ble-sdk #700: ACC Timestamp Alignment

Benjamin-Griffiths, 2025-09-16:
***Platform your question concerns:*** 
- [ ] Android 
- [ ] iOS
- [x] Other
- [ ] Platform is not relevant for this question

***Device:***
- [ ] Grit X2 Pro
- [ ] Vantage V3
- [ ] Vantage M3
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [x] Polar H10
- [ ] Polar H9
- [ ] Polar 360
- [ ] Other
- [ ] Device is not relevant for this question

**Description:**
I have a Python application that is parsing Polar H10 accelerometer data (using [this](https://github.com/polarofficial/polar-ble-sdk/blob/master/technical_documentation/online_measurement.pdf) documentation). However, it does not explicitly state whether the timestamp corresponds with the first sample in the array, or the last.

From [this](https://github.com/polarofficial/polar-ble-sdk/issues/189) issue, it's suggested that the timestamp corresponds to the last sample in the array, and you can back calculate the timestamps for previous samples.

Can I get confirmation that the timestamp is aligned with the last sample, and not the first?

Thanks!

samulimaa, 2025-09-16:
Hi,

We currently provide official support only for the native Android and iOS platforms.

From the older GitHub issue you linked: 

“as far as I understand the devices deliver the data batchwise whereby each batch has only one timestamp (which is the timestamp for the last data point in the batch).
… correct”

This should imply that the answer to your question is yes.

The official SDK currently returns an individual timestamp for every ACC sample.
You can check how the SDK handles ACC data and follow a similar approach.

Benjamin-Griffiths, 2025-09-18:
Thank you for the fast response!

Could you point me to where the timestamp per-sample calculation is performed in the SDK source code?

samulimaa, 2025-09-22:
Look up [PmdDataFrame](https://github.com/polarofficial/polar-ble-sdk/blob/244cf23d95de6750100eaa0ce26586631212424f/sources/Android/android-communications/library/src/main/java/com/polar/androidcommunications/api/ble/model/gatt/client/pmd/PmdDataFrame.kt#L6)/[AccData](https://github.com/polarofficial/polar-ble-sdk/blob/244cf23d95de6750100eaa0ce26586631212424f/sources/Android/android-communications/library/src/main/java/com/polar/androidcommunications/api/ble/model/gatt/client/pmd/model/AccData.kt) implementations.

There is also unit tests for Pmd/AccData available.

Benjamin-Griffiths, 2025-10-07:
Okay to close - thank you!

closed 2025-10-07 by Benjamin-Griffiths