polarofficial/polar-ble-sdk #702: Upgrading Polar 360 to latest firmware doesn't work starting offline ppi any more

abbjetmus, 2025-09-19:
**Platform on which you observed the bug:** 
- [ ] Android 
- [ ] iOS
- [ ] Other
- [ X] Platform is not relevant for this bug

**Platform version:**

**SDK version:**

**Customer app url**
Link to app store / play store / your website

**Device on which you observed the bug:**
- [ ] Grit X2 Pro
- [ ] Vantage V3
- [ ] Vantage M3
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ X] Polar 360
- [ ] Other
- [ ] Device is not relevant for this bug

**Describe the bug**
Description of your observation

We are using a Flutter wrapper to the polar sdk and works well but upgrading sensor firmware things has broken.

Calling startOfflineRecording with PPI doesn't work any more in firmware 3.X but it worked in 2.X.

I get errors:

```log
PlatformException(ERROR_STARTING_RECORDING, pmd cp command REQUEST_MEASUREMENT_START error: ERROR_INVALID_STATE, null, null)
```

This is a very urgent issue since we need it fixed before monday 22/9.

**How to Reproduce**
If you know how the bug may be reproduced please add the steps here

Use Polar 360 with firmware 3.X and call startOfflineRecording()

**Expected behavior**
Description of what you expected to happen. 

**Screenshots and logs**
If applicable, add screenshots or logs to help explain your problem.

This is the code that I'm using from Flutter -> Kotling calling the polar sdk.

```dart
          await polar.startOfflineRecording(
            sensorId,
            PolarDataType.ppi,
            settings: PolarSensorSetting(<PolarSettingType, int>{}),
          );

  Future<void> startOfflineRecording(
    String identifier,
    PolarDataType feature, {
    PolarSensorSetting? settings,
  }) async {
    await _methodChannel.invokeMethod(
      'startOfflineRecording',
      [
        identifier,
        feature.toJson(),
        settings != null ? jsonEncode(settings) : null,
      ],
    );
  }
```

```kt
    private fun startOfflineRecording(call: MethodCall, result: Result) {
        val arguments = call.arguments as List<*>
        val identifier = arguments[0] as String
        val feature = gson.fromJson(arguments[1] as String, PolarDeviceDataType::class.java)
        val settings = gson.fromJson(arguments[2] as String, PolarSensorSetting::class.java)

        wrapper.api
            .startOfflineRecording(identifier, feature, settings)
            .subscribe({
                runOnUiThread { result.success(null) }
            }, {
                runOnUiThread {
                    result.error("ERROR_STARTING_RECORDING", it.message, null)
                }
            })
            .discard()
    }
```

abPlus21, 2025-09-19:
Did you try to start the sensor without any settings, since this works and had worked across recent FW versions like this: 
APIHelper.api.startOfflineRecording(deviceId, dataType)

rkangast, 2025-09-22:
Hi @abbjetmus 

Before you start loading active time (or any other data) please use API [sendInitializationAndStartSyncNotifications](https://polarofficial.github.io/polar-ble-sdk/polar-sdk-android/com/polar/sdk/api/PolarBleApi.html#sendInitializationAndStartSyncNotifications(java.lang.String)). This will trigger flushing data  to the files in device. It will also request the device to go to faster transfer mode. Finally use the sendTerminateAndStopSyncNotifications API to inform the device to go back to slower transfer mode (to save battery life).

Also see [documentation](https://github.com/polarofficial/polar-ble-sdk/blob/master/documentation/SyncImplementationGuideline.md) (released last week) on implementing data synchronization on your app.

abbjetmus, 2025-09-22:
@rkangast Thanks! Just to clarify, here’s roughly the process I’m using:

1. polar.getOfflineRecordingStatus(sensorId)

2. polar.stopOfflineRecording(sensorId, PolarDataType.ppi)

3. polar.listOfflineRecordings(sensorId)

4. polar.getOfflinePpiRecord(sensorId, record)

5. polar.removeOfflineRecord(sensorId, record)

6. polar.startOfflineRecording(sensorId, PolarDataType.ppi)

7. polar.getActiveTime(sensorId, DateTime.now().startOfDay(), DateTime.now())

I have three questions about this flow and the 24/7 Activity API:

1. Initialization and termination calls
When should I call sendInitializationAndStartSyncNotifications(identifier) and sendTerminateAndStopSyncNotifications(identifier)?

Before step 1 and after step 6?

Before step 7 and after step 7?

Or only once before step 1 and after step 7?

2. No data from getActiveTime() after upgrading
After upgrading, getActiveTime() returns no data, whereas before it returned non-zero values:

[image]

Could this be because I didn’t call sendInitializationAndStartSyncNotifications()?
(It was working previously with firmware 2.X and SDK 6.1.0.)

3. Memory usage and data deletion
I’ve encountered MEMORY_FULL issues from users running the app for long periods. For offline data, we delete the records after reading. But for “active time” we never do.

Since we don’t explicitly start the Activity API (unlike OfflineRecordings), does the sensor keep tracking steps, distance, active time, calories, etc. by itself?

For Activity API data types we don’t use, do we still need to delete them?
Currently, in getActiveTime() we just update the values for the day, but we never call deleteStoredDeviceData(ACTIVITY). Could this be causing the memory issue?

closed 2025-09-22 by abbjetmus