polarofficial/polar-ble-sdk #705: getActivitySampleData gives me 'samples' is inaccessible due to 'internal' protection level`?

abbjetmus, 2025-09-28:
***Platform your question concerns:*** 
- [ ] Android 
- [ X] iOS
- [ ] Other
- [ ] Platform is not relevant for this question

**SDK version:**

**Customer app url**
Link to app store / play store / your website

***Device:***
- [ ] Grit X2 Pro
- [ ] Vantage V3
- [ ] Vantage M3
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ X] Polar 360
- [ ] Other
- [ ] Device is not relevant for this question

**Description:**
Hi! When calling getActivitySampleData on iOS with version 6.6.0 and trying to access the samples I get `'samples' is inaccessible due to 'internal' protection level`. I'm i missing something or maybe my XCode configuration? @rkangast 

activityData	PolarBleSdk.PolarActivityData	0x00000001572daf40
samples	PolarBleSdk.PolarActivityData.PolarActivitySamples?	some
startTime	Foundation.Date	2025-09-28 20:23:00 UTC	
metRecordingInterval	Int	30
metSamples	[Float]?	184 values	some
stepRecordingInterval	Int	60
stepSamples	[UInt32]?	91 values	some
activityInfoList	[PolarBleSdk.PolarActivityData.PolarActivityInfo]	16 values	

[image]

If I understand Swift correctly this means INTERNAL without specifying access modifier:

```swift
//  Copyright © 2025 Polar. All rights reserved.

import Foundation

// [PolarActivitySamplesData] data for given [startTime].
public class PolarActivityData: Codable {

    var samples: PolarActivitySamples?

    public struct PolarActivitySamples: Codable {
        public var startTime: Date
        public var metRecordingInterval: Int
        public var metSamples: [Float]!
        public var stepRecordingInterval: Int
        public var stepSamples: [UInt32]!
        public var activityInfoList: [PolarActivityInfo]
    }

    public struct PolarActivityInfo: Codable {
        public var activityClass: PolarActivityClass
        public var timeStamp: Date
        public var factor: Float
    }

    public enum PolarActivityClass: String, Codable {

        case SLEEP = "SLEEP"
        case SEDENTARY = "SEDENTARY"
        case LIGHT = "LIGHT"
        case CONTINUOUS_MODERATE = "CONTINUOUS_MODERATE"
        case INTERMITTENT_MODERATE = "INTERMITTENT_MODERATE"
        case CONTINUOUS_VIGOROUS = "CONTINUOUS_VIGOROUS"
        case INTERMITTENT_VIGOROUS = "INTERMITTENT_VIGOROUS"
        case NON_WEAR = "NON_WEAR"

        static func getByValue(value: Data_PbActivityInfo.ActivityClass) -> PolarActivityClass {

            switch value {

            case .sedentary:
                return .SEDENTARY
            case .sleep:
                return .SLEEP
            case .light:
                return .LIGHT
            case .nonWear:
                return .NON_WEAR
            case .continuousModerate:
                return .CONTINUOUS_MODERATE
            case .intermittentModerate:
                return .INTERMITTENT_MODERATE
            case .continuousVigorous:
                return .CONTINUOUS_VIGOROUS
            case .intermittentVigorous:
                return .INTERMITTENT_VIGOROUS
            }
        }
    }

    static func parsePbActivityInfoList(activityInfoList: [Data_PbActivityInfo]) throws -> [PolarActivityInfo] {
        var polarActivityInfoList: [PolarActivityData.PolarActivityInfo] = []
        do {
            for activityInfo in activityInfoList {
                polarActivityInfoList.append(
                    PolarActivityData.PolarActivityInfo(
                        activityClass: PolarActivityData.PolarActivityClass.getByValue(value: activityInfo.value),
                        timeStamp: try PolarTimeUtils.pbLocalDateTimeToDate(pbLocalDateTime: activityInfo.timeStamp),
                        factor: activityInfo.factor
                    )
                )
            }
        } catch {
            throw PolarErrors.dateTimeFormatFailed()
        }
        return polarActivityInfoList
    }
}

public struct PolarActivityDayData: Codable {
    public var polarActivityDataList: [PolarActivityData]
}

```

rkangast, 2025-09-29:
Hi @abbjetmus,

You're correct, the field 'samples' should be marked public. This will be fixed in the next SDK release (6.8.0).

Our internal ticket numberis TC-66007.

abbjetmus, 2025-09-29:
@rkangast I'm not sure deleteDevicesDateFolders is working on iOS either, I call it and it succeeds but the data is not removed. I pass dates from now to a week back in time. And I still get yesterdays data.

```swift
    _ = api.deleteDeviceDateFolders(identifier, fromDate: fromDate, toDate: toDate)
        .subscribe(
            onCompleted: {
                result(nil)
            },
            onError: { error in
                result(FlutterError(code: "ERROR_DELETING_FOLDERS",
                                  message: error.localizedDescription,
                                  details: nil))
            }
        )
```

```log
PolarActivityInfo(timeStamp: 2025-09-28 22:10:30.000Z, activityClass: SEDENTARY, factor: 1.0), PolarActivityInfo(timeStamp: 2025-09-28 22:25:30.000Z, activityClass: SLEEP, factor: 1.0), PolarActivityInfo(timeStamp: 2025-09-28 22:29:00.000Z, activityClass: NON_WEAR, factor: 1.0), PolarActivityInfo(timeStamp: 2025-09-29 05:34:00.000Z, activityClass: SLEEP, factor: 1.0), PolarActivityInfo(timeStamp: 2025-09-29 05:34:30.000Z, activityClass: NON_WEAR, factor: 1.0), PolarActivityInfo(timeStamp: 2025-09-29 05:40:30.000Z, activityClass: LIGHT, factor: 1.0), PolarActivityInfo(timeStamp: 2025-09-29 05:47:00.000Z, activityClass: SLEEP, factor: 1.0), PolarActivityInfo(timeStamp: 2025-09-29 05:48:00.000Z, activityClass: LIGHT, factor: 1.0), PolarActivityInfo(timeStamp: 2025-09-29 05:54:00.000Z, activityClass: SEDENTARY, factor: 1.0), PolarActivityInfo(timeStamp: 2025-09-29 05:57:30.000Z, activityClass: LIGHT, factor: 1.0), PolarActivityInfo(timeStamp: 2025-09-29 06:00:30.000Z, activityClass: SEDENTARY, factor: 1.0), PolarActivityInfo(timeStamp: 2025-09-29 06:02:00.000Z, activityClass: LIGHT, factor: 1.0), PolarActivityInfo(timeStamp: 2025-09-29 06:08:00.000Z, activityClass: SEDENTARY, factor: 1.0), PolarActivityInfo(timeStamp: 2025-09-29 06:11:30.000Z, activityClass: LIGHT, factor: 1.0), PolarActivityInfo(timeStamp: 2025-09-29 06:24:00.000Z, activityClass: SEDENTARY, factor: 1.0), PolarActivityInfo(timeStamp: 2025-09-29 06:37:00.000Z, activityClass: LIGHT, factor: 1.0), PolarActivityInfo(timeStamp: 2025-09-29 06:38:30.000Z, activityClass: CONTINUOUS_MODERATE, factor: 1.0), PolarActivityInfo(timeStamp: 2025-09-29 06:39:30.000Z, activityClass: LIGHT, factor: 1.0), PolarActivityInfo(timeStamp: 2025-09-29 06:55:00.000Z, activityClass: SEDENTARY, factor: 1.0), PolarActivityInfo(timeStamp: 2025-09-29 06:56:30.000Z, activityClass: LIGHT, factor: 1.0), PolarActivityInfo(timeStamp: 2025-09-29 06:58:30.000Z, activityClass: SEDENTARY, factor: 1.0)]```

abbjetmus, 2025-09-29:
@rkangast I'm on firmware 3.X. I feel that I have tried formatting dates as much as I can but still doesn't work.

If you look at the example my code is passing 2025-09-29 but the date formatter is turning it to UTC when calling the sensor.

[image]

The same goes for:
```swift
func deleteStoredDeviceData(_ identifier: String, dataType: [PolarStoredDataType](https://polarofficial.github.io/polar-ble-sdk/polar-sdk-ios/Classes/PolarStoredDataType.html).StoredDataType, until: Date?) -> Completable
```

Calling both of these and the getActiveTime we get the whole day of activeTime, so it is not deleted!

palmqvisti, 2025-10-07:
The bug will be fixed in the next SDK release.